Yogui79/IntexPureSpa #17: SearchNetworkID detection time

fgsam1234, 2021-07-12:
### Discussed in https://github.com/Yogui79/IntexPureSpa/discussions/16

<div type='discussions-op-text'>

<sup>Originally posted by **fgsam1234** July  9, 2021</sup>

Hi everyone,

i am running the SearchNetworkID scetch for 7 days and still have no hit. Could there be an issue or can this take so long?
I remember typically 36 hours being mentioned, which is reasonable if you consider 17^4 options for the NetworkID and 1 second per test.
It is a  28462 PureSpa.
The channel has been successfully detected with the fast blinking LED (0x4B). With commented //#define SEARCH_NETWORK_ID the LC12 LED is continuously blinking fast. 
When the SearchNetworkID scetch is running with uncommented #define SEARCH_NETWORK_ID, Putty is displaying each second a new line like the one below:

Config AA 5A B9 46 07 94 00 03 00 04 00 4B 00 00 00 12 00 02
AA 5B 9F 70 07 94 00 03 00 04 00 4B 00 00 00 12 00 13

Is this normal and i have to wait for some additional days, or could something be going wrong.

@Yogui79 or @LEECHER1 could you please help me?

Kind regards,
fgsam1234

PS: A strange thing i have noticed is that the ID should be the block 5 and 6 of the hex code - i guess. - but the sketch has already send all codes from 0000 to FFFF and is now continuing from the start once again, just now the block 18 is different then before.</div>

Yogui79, 2021-07-12:
Hello

First of all you don't need to post 2 time the same post
Second as mentioned this code is delivered without warranty 
I normally I answered quickly but I do it in my free time and actually I haven't free time to analyse your problem

The log you provide looks good and for me you should find the network id 
As you see the code only increase the network id and wait until the spa answered if not he increase again.

Could you try the next channel or the channel before.
Please provide a complete log file.

fgsam1234, 2021-07-13:
Dear Yogui79,

Thank you for analyzing my issue. I have attached the putty log to this post. 
Could you please delete the post under discussions. I just used the the button "generate issue from this post" unfortunately it just copied it to Issues and did not delete the initial post in discussions.

I have checked the neighbor channels with the  SearchNetworkID sketch with commented //#define SEARCH_NETWORK_ID. 0x4A does not produce any fast flashing of the LC12 LED. 0x4C produces a fast flashing LED, if i am close to the pump. If I am more then 2 meters away from the pump, it does not flash fast anymore.

I will now try 0x4C, just to be sure.

[putty.log](https://github.com/Yogui79/IntexPureSpa/files/6807056/putty.log)

Yogui79, 2021-07-15:
ok it means 0x4B is the right chanel
please be so close as possible from the pump for searching the network id. the remote control should also be on the pump because in this case the pump sent more that 5 signal/sec if you remove the control from the pump it send only 1/sec

fgsam1234, 2021-07-15:
Ah thanks, that is a good hint. I had the remote detatched.
The LC12 is about 0.3 meters away from the pump. So the distance should be OK.
I will try again with Channel 0x4B and the remote lying on the charging slot of the pump.

fgsam1234, 2021-07-17:
Just for my understanding: 
The SearchNetworkID is trying all networkIDs from 0000 to FFFF. It is calculating the checksum from each, sends it and is waiting 1 second for the pump to answer correctly. If not positive, it tries the next.
So if block 5 an 6 of the checksum have reached FF FF all possible combinations gave been tested, correct?

Does it make sense to continue the sketch once FFFF had been reached and no networkID has been found?

fgsam1234, 2021-07-17:
Unfortunately the networkID sketch has reached FFFF for block 5 and 6.
It is still continuing to test IDs. It is currently once again at 0445.
Is it senseful to let it continue? or should I stop it and try something different? If yes, what could I try.

fgsam1234, 2021-07-19:
I was finally able to detect the correct Channel and Network ID. :-) 

For that I have changed the delay in the Channelsearch sketch to 3000ms and in the networkID for sketchLastTimeReciveData to 1500ms. By that the Channel could be clearly identified and I had a hit with the networkID within roughly 8 hours.

It works now very well. 

Thank you @Yogui79 for the code itself and your support.

closed 2021-07-20 by Yogui79

MathieuJouslin, 2022-07-03:
> I was finally able to detect the correct Channel and Network ID. :-)
> 
> For that I have changed the delay in the Channelsearch sketch to 3000ms and the delay in the networkID sketch for LastTimeReciveData to 1500ms. By that the Channel could be clearly identified and I had a hit with the networkID within roughly 8 hours.
> 
> It works now very well.
> 
> Thank you @Yogui79 for the code itself and your support.

Hello, where did you changed the delay, in the loop function or in the SetSettingsChannel function ?