Yogui79/IntexPureSpa #26: Can't find channel

EricVdB, 2021-07-30:
Hi,

first of all this looks really awesome, it would finally allow me to integrate the SPA into our home automation using OpenHab.

But I can't get the channel search to work. It keeps on blinking steady and never blinks fast. I tried close to the SPA pump (like 5cm away), 1 meter further away, no joy.

If I understood the code well, the fast blinking should happen after you set the channel and the led goes back off, right? (DO_SET = LOW at line 134 in SearchChannel.ino)

Is there anything else i can try?

Model of the SPA = 28442 (bought in belgium)

Kind regards,
Eric

djgel, 2021-07-30:
Hi,  other  user got  a  same  problem. https://github.com/Yogui79/IntexPureSpa/discussions/7 

Another   question : Using  a  ESP32 V4  dev model ?

"If I understood the code well, the fast blinking should happen after you set the channel and the led goes back off, right? (DO_SET = LOW at line 134 in SearchChannel.ino)" No, Light turn ON in setting  channel and  turn OFF because  not  rceived data. 

Try use correct  board and  review  conections!

EricVdB, 2021-08-02:
Don't have a v4 board available, so I tried with a Arduino, still same result, the fast blinking never happens.
I attached my scope on the solderings and connections are good. I can decode the HEX on the serial line.

Any other ideas?

Yogui79, 2021-08-17:
hello @EricVdB  
please post the serial connection logger

bendiger, 2022-04-13:
I have the same issue, I have a hot tub thats not one of the confirmed models thought. All other hardware is the same as whats listed. I have model 28441E the 6 person. I have run through the channel search about 10 times, at varying distance from the pump and from the controller. Still no fast blinks. Not sure what I'm missing, I really hope I don't have a newer model that isn't compatible with this...

bendiger, 2022-04-13:
Shoot.... so I guess my version is the 2021 model. Any hope of getting that one to work with this? Is there anything I can do to help? I do that two of them because of an RMA.

LEECHER1, 2022-04-13:
@bendig
It is easy to check the Spa version. See the screenshot.
Do you have 2021. Is WiFi default than can you use the INTEX APP.
 
[​https://ibb.co/LZPTPRZ](https://ibb.co/LZPTPRZ)
(Version 2020)

https://ibb.co/b6kX9Sk
(Version 2021 - with WIFI symbol in the left corner)

We haven't tried to works with the 2021, I think it's possible.

Please see how @djgel already posted that the ESP is the v4 developer version.

bertdeelman, 2022-06-09:
Hi,

I can't also retrieve the Channel and I have a 2021 model Spa. Any ideas how the solve this? And is there a way to easy check if I have the right board? 

Regards,
Bert

PLCHome, 2023-03-04:
I also have a model from 2021. Before that I had the old one without WiFi. I couldn't find the channel either. However, I noticed that the small blue LED behind the display flashed faster in the old pool. I pressed the lock button to search for a channel. Then you can see the blue LED flashing behind the display. Then the LC12 also flashed at 0x2b.