Yogui79/IntexPureSpa #38: Home assistant: MQTT Mosquito LOG =>Look like channel change since last boot

egenu, 2021-10-31:
I followed the procedure to find the channel ID and then the Network ID, in my case 0x4C and 0xFFF
I am Fighting to setup my IntexPureSpa, control, with Home assistant: MQTT Mosquito.
I found in my MQTT logs very strange messages:
1635709111: New connection from 192.168.1.144 on port 1883.
1635709111: New client connected from 192.168.1.144 as IntexSpa (p2, c1, k15).
1635709128: New connection from 172.30.32.2 on port 1883.
1635709128: Socket error on client <unknown>, disconnecting.
1635709154: Client IntexSpa has exceeded timeout, disconnecting.
1635709248: New connection from 172.30.32.2 on port 1883.
1635709248: Socket error on client <unknown>, disconnecting.
	a
Could someone give me a hint to search in the right direction?

Yogui79, 2021-11-21:
@LEECHER1  as you use Home Assistna could you help ?

but looks like the IP dresse is wrong configurated

LEECHER1, 2021-11-21:
@egenu 
Hey,

It looks the ESP32 has a bad Network (Wifi) connection. 

Is the IP 192.168.1.144 the ESP32? 
Please, fix the IP on the router of the ESP as static IP.

Check the WIFI Siginal with this code: 
https://learn.adafruit.com/adafruit-metro-esp32-s2/wifi-test

Do you see on this link what dBM values say:
https://eyenetworks.no/en/wifi-signal-strength/

br LEECHER1

egenu, 2021-11-25:
Hi  LEECHER1 and Yogui79,

First of all I would like to say that I am amazed about the work that you have done and you have all my respect for such work!

The IP 192.168.1.144 is the ESP32, I fixed it as static IP on the router, thanks for your suggestion.
I made the Wifi test and I got a -48 dBm signal, then I believe this is not the issue.

On the other hand I retry the full procedure many times and the Channel ID has change 3x during my successive attempt , is this normal?

The last attempt that I made after applying your suggestions give me a 0x9C channel who is the first time and the Network ID program is not scanning anymore like it was the first times I launched, maybe due to Arduino updates that I had to apply?

Could someone please confirm that the SearchNetworkId.ino program is still working with latest Arduino program and library updates?

Thanks again for sharing and giving all that work to the comunity!
Egenu

Yogui79, 2021-11-26:
hello 

_"On the other hand I retry the full procedure many times and the Channel ID has change 3x during my successive attempt , is this normal?_"

-> yes it's possible

please give me version you use for :+1: 
Arduino ide you use
the esp32 lib you use

egenu, 2021-11-27:
Hello

Actually I have installed Arduino IDE    1.8.16
and  I use esp32 by Expressif Systems   1.0.6
EspMQTTClient              1.13.2
arduino-timer                2.3.0
ArduinoOTA                    1.0.7

Hope this was you where looking for.

Yogui79, 2021-11-27:
I haven't test with arduino ide 1.8.16 but should work
 esp32 by Expressif Systems 1.0.6 I use alsa this one.

the SearchNetworkId.ino  sketch is only to detect the network id from the LC12s from the pump and the first preoblem you speek about was a mqtt pissue.

please could you be more clear : 
1 do you find the chanel ?
2 download the sketch  SearchNetworkId.ino and modifie the line uint8_t Channel =0x47; with your chanel
3 post a video from your LC12s after that

egenu, 2021-11-27:
Hello, sorry to add more confusion than wanted. Due to the fact that I didn't understood, if my issue was coming from the fact that the source of my unsuccessful attempt was coming from the :
MQTT <=> Home assistant or SPA <=> Board with the channel ID and then the Network ID found a
Then I retry many times the full operation and this is why I was surprised to see a different chanel.

To answer to your questions
1) do you find the chanel ?
-> Yes 9B & 9C
2) Edited the line uint8_t Channel =0x9B ;and download it
[image]

3) -> https://user-images.githubusercontent.com/73351718/143678705-afa00b44-c4f3-4772-9b33-0106038f859d.MOV

Yogui79, 2021-11-27:
look's good
now remove the // from this line #define SEARCH_NETWORK_ID
upload the sketch 
start monitoring the ouput with puty and save the data in a file the search network id  can last up to 36 hours you will ahve a message when the network id is found

egenu, 2021-11-28:
Hi,

Look like my previous NETWORK_ID didn't success, even if I had a similar message at the end.
----------------------------
  Your network id is:
         0x153F
----------------------------
Then I put the "used_chanel" (0x9B) and "used_network" (0x153F) on my "PureSpa Sofware" with the MQTT and WIFI settings
and sudainly works like a charm. 

[image]

Many thanks to both.

egenu, 2021-11-28:
Still an issue :(
after removing the PC and using a wall mount plug the board is blinking like before, but I am unable to get control / communication with Homeassistant, but the fixe IP on the board is answering his the IP 192.168.1.144 (ping)
what did I do wrong!
Cheers

LEECHER1, 2021-11-28:
Is the communication with ESP32 to Pump and Controller okay?
Maybe the channel has changed. Is possible from time to time. Wait a little bit ~10min

Or is only the communication with ESP32 to Home Assistant not okay?

egenu, 2021-11-28:
I have trouble to understand, how to diagnose the communication with ESP32 to Pump and Controller? Is only with the blinking led that you are able know? How to access the serial communication when the ESP32 is not anymore USB connected to my PC, with OTA Arduino IDE complains that the:
 "Serial Monitor is not supported on network ports such as 192.168.1.144 for the ESP32 Dev Module i this release"

From the Home Assistant Mosquito MQTT broker log I believe it works:
I see the following, and I assume this parts works, but as it do not update he is in TIMEOUT situation:

1638114518: New connection from 192.168.1.144 on port 1883.
1638114518: New client connected from 192.168.1.144 as IntexSpa (p2, c1, k15).
...
1638114978: Client IntexSpa has exceeded timeout, disconnecting.

Yogui79, 2021-11-30:
` have trouble to understand, how to diagnose the communication with ESP32 to Pump and Controller? Is only with the blinking led that you are able know? ` yes only over the blinking from the blue led it should be very fast
yo have no way to diagnose over ota

egenu, 2021-12-04:
Hello good news,

I had to reset completely my SPA. the controller "tablet" discharging it completely .
I left the SPA without power for 24 hours to.
After this "RESET", I just try the connectivity using a wall mount plug for the ESP32 board.
He is blinking like before, and everything start working inside home assistant after 10 minutes like you said!
Inside the MQTT logs I saw the INTEXSPA connectivity and doesn't timeout anymore.
A big thanks to both and if you have an By me a Beer or something else I would be happy to contribute.

closed 2021-12-04 by egenu