Yogui79/IntexPureSpa #39: Spa Programm not working

HansEhls, 2021-11-18:
Hi there,
I have an Intex Pure Spa 28462 whirlpool and am trying to get the GitHub project up and running.

Hardware built wasn't that difficult. 
Connected, I also found the channel, then also the network address.

Now I have loaded the spa program and I can't get any further because unfortunately I have no knowledge of programming.

The program logs on to the network and also to the MQTT, then some values ​​(probably the different statuses) are communicated wildly between the pump and the program. Then it breaks off and looks for a new channel and the whole thing starts over.
I switched on all debugs and downloaded the serial info.

Could someone help me and try to understand what the problem is?

The program also logs on to Mosquitto and sends up 2 types of messages, but none such as temperature or similar.

I also installed Home Assistant in case a handshake was required. These are the CMD messages that you see in the Mosqutitto.

It would be great if someone could help me. 

Thanks in advance

[image]
[Serial Info.odt](https://github.com/Yogui79/IntexPureSpa/files/7565357/Serial.Info.odt)

w00dst0ck, 2021-11-18:
Have you tried to reduce the distance between the Intex and the ESP32. That fixed the problem with the lost channel for me.

HansEhls, 2021-11-18:
Hi, 
thanx for the tip, but no change, even when it direct on the pump

LEECHER1, 2021-11-19:
Please try to reset the Pumpe and the distance ~1m and wait 15 min. > Not working try again.

HansEhls, 2021-11-19:
Not working. Same serial protokoll

Yogui79, 2021-11-21:
regarding the log you have distubation during communication between pump and ESP looks like you only communicate with remote control.
-> switch ESP off
->switch pump off (unplug)
->switch remoter control off
->remove remote control from the pump

-> switch on pump and reset it
-> put remote control on pump
-> be sure you dont have E81
-> switch on ESP

if not working please send a log configurate like this : 

//Uncomment following line to have more debug infos
//#define DEBUG_RECIEVED_DATA
#define DEBUG_SEARCH_CHANNEL
#define DEBUG_SEND_COMMAND
#define DEBUG_PUMP_DATA
#define DEBUG_CONTROLLER_DATA
#define DEBUG_CONFIG
//#define DEBUG_MQTT
//#define DEBUG_SEND_VALUE_TO_HOME_AUTOMATION_SW

HansEhls, 2021-11-22:
Hi,

first of all thank you, that you are trying to help. no change and as I removed some debug data, the serial info is smaller now.

20:38:44.711 -> Used first command read from EEPROM 0x4B
20:38:44.711 ->  Config AA 5A B9 46 FF FF 00 00 00 04 00 4B 00 00 00 12 00 62 
20:38:51.032 -> Debug Send command : 4B 00 00 00 00 00 A9 B2  
20:38:52.629 -> Debug Send command : 4B 00 00 00 00 00 A9 B2  
20:38:54.233 -> Debug Send command : 4B 00 00 00 00 00 A9 B2  
20:38:55.462 -> Look like channel change since last boot
20:38:55.462 ->  Config AA 5A B9 46 FF FF 00 00 00 04 00 41 00 00 00 12 00 58 
20:38:56.955 -> 
20:38:56.955 -> Debug search Chanel : AA 5B D6 F5 FF FF 00 00 00 04 00 41 00 00 00 12 00 25  Config AA 5A B9 46 FF FF 00 00 00 04 00 42 00 00 00 12 00 59 
20:38:59.432 -> 
20:38:59.432 -> Debug search Chanel : AA 5B D6 F5 FF FF 00 00 00 04 00 42 00 00 00 12 00 26  Config AA 5A B9 46 FF FF 00 00 00 04 00 43 00 00 00 12 00 5A

Yogui79, 2021-11-22:
please post the debug data after the esp start 
somme important information for me are her mentioned

HansEhls, 2021-11-22:
22:07:02.488 -> Ready
22:07:02.488 -> IP address: 192.168.178.201
22:07:03.579 -> Used Channel read from EEPROM 0x4B
22:07:03.579 -> Used first command read from EEPROM 0x4B
22:07:03.579 ->  Config AA 5A B9 46 FF FF 00 00 00 04 00 4B 00 00 00 12 00 62 
22:07:05.078 -> AA IntexSpa/Communication with pump 1
22:07:05.882 -> 5B IntexSpa/Communication with pump 1
22:07:05.916 -> D6 IntexSpa/Communication with pump 1
22:07:05.916 -> F5 IntexSpa/Communication with pump 1
22:07:05.916 -> FF IntexSpa/Communication with pump 1
22:07:05.916 -> FF IntexSpa/Communication with pump 1
22:07:05.916 -> 00 IntexSpa/Communication with pump 1

HansEhls, 2021-11-22:
this comes up, when I reset the buttons on the board

22:48:02.289 -> ets Jun  8 2016 00:22:57
22:48:02.289 -> 
22:48:02.289 -> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
22:48:02.289 -> waiting for download
22:48:03.508 -> ets Jun  8 2016 00:22:57
22:48:03.508 -> 
22:48:03.508 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:48:03.508 -> configsip: 0, SPIWP:0xee
22:48:03.545 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:48:03.545 -> mode:DIO, clock div:1
22:48:03.545 -> load:0x3fff0018,len:4
22:48:03.545 -> load:0x3fff001c,len:1216
22:48:03.545 -> ho 0 tail 12 room 4
22:48:03.545 -> load:0x40078000,len:10944
22:48:03.545 -> load:0x40080400,len:6388
22:48:03.545 -> entry 0x400806b4
22:48:03.789 -> E (64) psram: PSRAM ID read error: 0xffffffff
22:48:03.823 -> ----------------------------------------------------------------------------------------------------------------------
22:48:03.860 -> 
22:48:03.860 ->           Intex Purespa communication sketch version 1.1.1 build on : 2021.07.31 21:27:53
22:48:03.860 -> 
22:48:03.860 -> ----------------------------------------------------------------------------------------------------------------------
22:48:03.860 -> Booting
22:48:06.167 -> Ready
22:48:06.167 -> IP address: 192.168.178.201
22:48:07.291 -> Used Channel read from EEPROM 0x4B
22:48:07.291 -> Used first command read from EEPROM 0x4B
22:48:07.291 ->  Config AA 5A B9 46 FF FF 00 00 00 04 00 4B 00 00 00 12 00 62 
22:48:13.661 -> Debug Send command : 4B 00 00 00 00 00 A9 B2  
22:48:15.657 -> Debug Send command : 4B 00 00 00 00 00 A9 B2  
22:48:19.305 -> Debug Send command : 4B 00 00 00 00 00 A9 B2  
22:48:19.305 -> Look like channel change since last boot
22:48:19.305 ->  Config AA 5A B9 46 FF FF 00 00 00 04 00 41 00 00 00 12 00 58 
22:48:20.813 -> 03 FF 00

HansEhls, 2021-11-22:
[image]
I do not know, if that is important, but in the edge down, right, there is a blue LED which is blinking fast all the time. I have no idea, if that is normal

Yogui79, 2021-11-24:
this is norma this is the conection between pum an remote control
how long did you wait after starting your ESP

HansEhls, 2021-11-24:
in the meanwhile several times over 30 min.

Yogui79, 2021-11-26:
could you poste your project ?

HansEhls, 2021-11-27:
I copied the Spa programm and did only these changes!

#define USED_NETWORK_ID    0xFFFF
#define USED_CHANNEL       **0x4B**

uncomment "//" #define _28458_28462_   // For spa model #28458 #28462 #28457(US) #28461(US)
//#define _28442_28440_   // For spa Model #28442 #28440

#if  defined (_28458_28462_) &&  defined (_28442_28440_)
  #error select only one SPA model
#endif
#if  !defined (_28458_28462_) &&  !defined (_28442_28440_)
  #error select a SPA model
#endif

/*******************************************************
*
*   S E L E C T  T H E   P R O T O C O L   Y O U   U S E
*  
********************************************************/
//#define _MY_SENSORS_
#define _MQTT_

#ifdef ESP32
const char* Myssid = **"My Wlan"**;
const char* Mypassword = **"My Pass"**;
#endif

EspMQTTClient client(
  Myssid,
  Mypassword,
  "192.168.178.97",  // MQTT Broker server ip
  **"Iobroker"**,   // Can be omitted if not needed
  **"Password",**   // Can be omitted if not needed
  "IntexSpa",     // Client name that uniquely identify your device
  1883              // The MQTT port, default to 1883. this line can be omitted
);
#endif

Yogui79, 2021-11-27:
coul you olease donwload the SearchNetworkId.ino sketch 
make following changes

uint8_t Channel =0x4B; 
#define SEARCH_NETWORK_ID
uint16_t idtosend =0xFFFA;

en postet the serial monitor  after 10 mins

HansEhls, 2021-11-28:
I did a new channel search and found E1. Then I did a new network search and found 0x14f7. With that in the Spa Programm it did connect to the pump and also wrote the data one time into MQTT broker. 

BUT:
When the ESP was connecting to the pump the controller went do E81 error. when I put off the ESP the controller searched the pump end reconnected. I did that several time, but had always the same result. I did many resets of the pump to try that again, with the same result. I think after 5 resets the ESP does not find the pump anymore. 

So I starting over, reseted the pump, started the controller. After that connection seams now stable, i searched the channel, which is back to 4B. then I started the Nework scan with 0x14F0 so see, if it will find again 0x14F7. That didn´t work. So now I started the network search at 0x0000 again and wait, if it will find another network address.

Did I do something wrong?

THanks for your help. I really appreciate that.

BR
Hans

Yogui79, 2021-11-30:
Hi, 
we newer se a networkid  who change but afrer switching off the pump it could be that the chanel change

HansEhls, 2021-12-01:
Update!!!
I had a connection to the controller and the pump with 0xFFFF abd 0x4B again. I played around with some own debugs to understand the programm. As I said, I am completely new to the topic programming. 

Then somehow I lost the connection again and had to do a channel and network search again. I found the channel 0xA4 and with that again the netword 0x14F7. 

With that, at least for the moment (the pump is in my office at the moment) I have the connection established and the programm is running. 

I also have the data in MQTT, so I can start to look for the VIS in Iobroker. 

Hopefully the program will work also in the future, when I setup the pool again outside in the next days. 

Thanks again for your help and the effort you spend on that!

BR
Hans

closed 2021-12-01 by HansEhls