Yogui79/IntexPureSpa #40: Network ID search

delgigio, 2021-11-19:
Good morning,
I have carefully followed the guide you have given me, I have managed to find the channels with which the whirlpool communicates which are 12 and 13.
Unfortunately, when I load the program to search for the Network ID, the blue LED does not flash as indicated, it always has the same frequency as all the other channels. 
The detection of the channels has been done over and over again and the result was always channel 12 and 13. I have also tried inserting channels 9 10 11 12 13 14 15 16 but with very bad results.
What can I do? 
I hope you can help me... Thank you in advance.

HansEhls, 2021-11-19:
Could you Upload a screenshot of the Programm when you entered the channel

delgigio, 2021-11-21:
This is the channel search program

/*
 * THIS SOFTWARE IS PROVIDED "AS IS" AND
 * ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
 * WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL <COPYRIGHT HOLDER> BE LIABLE FOR ANY
 * DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES 
 * (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
 * LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND
 * ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
 * (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
 * SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 *
 *   Pin connection
 *                      
 *                       LC12S     Arduino     ESP
 *  ___________________
 * |                  o| GNG       GND         GND
 * |                  o| CS        D5          D18
 * |                  o| SET       D6          D19
 * |                  o| TX        D2          D16
 * |                  o| RX        D4          D17
 * |__________________o| VCC       3.3V        3.3V
*/

uint8_t test= 0x00;

#if defined (__AVR__) 
#include <SoftwareSerial.h>
SoftwareSerial mySerial(2, 4); // RX, TX
// Output defines
#define DO_SET    6
#define DO_CS     5
#elif defined (ESP32)
#define mySerial   Serial2
#define DO_SET    19
#define DO_CS     18
#endif

void setup() {
  mySerial.begin(9600);
  Serial.begin(115200);

  pinMode(DO_SET, OUTPUT);        // SET
  pinMode(DO_CS, OUTPUT);        // CS - можно просто соединить с GND
  delay(1000);
  
  digitalWrite(DO_CS, LOW);     // притягиваем к массе 
  digitalWrite(DO_SET, LOW);
  delay(100);

 // querryversion();

}

void loop() {

  SetSettingsChannel(test++);
  Serial.print("Channel "); 
  Serial.println(test-1, HEX);
  delay(1000);
}

void SetSettingsChannel(uint8_t channelid){
  byte Config[20];
  char res[5];
  
  
  Config[1]=0xAA;  
[... 311 lines cut ...]
  Config[15]=0x00;
  Config[16]=0x00;
  Config[17]=0x00;
  Config[18] =0x07;
  
  Serial.print (F("Querry version "));
  for(int i=1;i<19;i++){
    sprintf(&res[0],"%02X",Config[i]);
    Serial.print(res);
    Serial.print(" ");    
  }

  Serial.println (F(""));
  digitalWrite(DO_SET, LOW);
  delay(500);

  for(int i=1;i<19;i++){

    mySerial.write(Config[i]);
  }
 delay(1000);
 digitalWrite(DO_SET, HIGH);
 
 
}

Yogui79, 2021-11-21:
Hello,

1. could you please say me whats append if you comment this line  
`//#define SEARCH_NETWORK_ID`

2.Enable again this line 
`#define SEARCH_NETWORK_ID`

3. Use putty an log the serial communication, as mentioned in the description, at lease 36h 
4. Poste the logged file

delgigio, 2021-11-21:
Evening,
point one I did not understand what you mean.
For point two should I add the two bars? // or what should I write?
Point three the Putty I haven't used yet since the led light doesn't have the correct frequency.

Thanks for helping me

Yogui79, 2021-11-21:
for point 1 make a video please from the LC12S with the right chanel

for point 2 write #define SEARCH_NETWORK_ID  on line 28 from the SearchNetworkId.ino

Yogui79, 2021-11-21:
otther remarke you need to be close to your pump 
pump and display should be on without error E81

delgigio, 2021-11-22:
To do the signal detection I was close to about 50 centimetres with the display on without any error. 
In the program loaded on ESP32 I had written as you indicated (#define SEARCH_NETWORK_ID).
Thank you