Yogui79/IntexPureSpa #44: configuration.yaml does not parse properly for HomeAssistant

TaosEnergy, 2022-03-05:
From line 149 down (#Template Sensor) the file does not validate/ does not parse properly.  There is some sort of syntax issue(s).  I used codebeautify.org/yaml-validator

LEECHER1, 2022-03-06:
Hey,

Thx i will check it, but the YAML files is only an example.

LEECHER1, 2022-07-02:
> The way MQTT-Entities are managed have changed since one of the last
> Home-Asstitant-Updates. I made a MQTT-Configuration myself and would like
> to share it with you. Feel free to use it:
>
> I added this to my configuration.yaml
> mqtt: !include mqtt.yaml
>
> i added a mqtt.yaml-file with this content:
>
> # Pool Heizungssteuerung
>
> climate:
>
> - unique_id: intex_pool_hvac
>
>   name: "Intex Pool"
>
>
>
>   # Gerät
>
>   device:
>
>     identifiers:
>
> [... 404 lines cut ...]
>     payload_on: "1"
>
>     payload_off: "0"
>
>     device_class: connectivity

Thanks :)

what has changed?

guevara777, 2022-07-04:
[image]

That has changed ;) -> https://www.home-assistant.io/blog/2022/06/01/release-20226/ -> See "Breaking Changes"

madhehler, 2023-04-28:
Hello
I am really struggling to integrate the ESP into my HA
Can you please help me, I really getting crazy

Greetinsg
Mat

madhehler, 2023-04-30:
Hello

can you please provide me the whole configuration then, I am really lost

rommess, 2023-06-13:
Hi guys, i'de like to share my "code" in the new format, as i seen many poeple having trouble. based on @guevara777 code.
work for me so far.
@Portos1327 

`
mqtt:

  # Intex Pool ESP-Restart Button
  button:
  - unique_id: intex_pool_restart
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd Reset ESP
    payload_press: "reset"
    name: "Intex Pool Restart"
    entity_category: "config"
    device_class: "restart"
  
  # Intex Pool Decrease SetTemp Button
  - unique_id: intex_pool_decrease
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd decrease
    payload_press: "1"
    name: "Intex Pool Set Temp Decrease"
  
  # Intex Pool Increase SetTemp Button
  - unique_id: intex_pool_increase
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd increase
    payload_press: "1"
    name: "Intex Pool Set Temp Increase"
  
  # Select
  select:
  
  # Intex Pool Filter Time Input Select
  - unique_id: intex_pool_filter_setup_time_select
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd water filter time
    state_topic: IntexSpa/filter setup time
    name: "Intex Pool Filter Setup Time Select"
    options:
#      - "0"
      - "2"
      - "4"
      - "6"
  
  # Intex Pool Sanitizer Time Input Select
  - unique_id: intex_pool_sanitizer_setup_time_select
    device:
        identifiers:
          - "Intex Spa"
        manufacturer: "Intex"
        model: "SPA"
        name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd Sanitizer time
    state_topic: IntexSpa/Sanitizer setup time
    name: "Intex Pool Sanitizer Setup Time Select"
    options:
#        - "0"
        - "3"
        - "5"
        - "8"
  
  # MQTT Pool Schalter
  switch:
  
  - unique_id: intex_pool_power_switch
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd Power on off
    state_topic: IntexSpa/Power on
    payload_on: "1"
    payload_off: "0"
    name: "Intex Pool Power"
    
  - unique_id: intex_pool_filter_switch
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd water filter on off
    state_topic: IntexSpa/filter on
    payload_on: "1"
    payload_off: "0"
    name: "Intex Pool Filter"
    
  - unique_id: intex_pool_bubble_switch
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd bubble on off
    state_topic: IntexSpa/Bubble on
    payload_on: "1"
    payload_off: "0"
    name: "Intex Pool Bubble"
    
  - unique_id: intex_pool_heater_switch
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd heater on off
    state_topic: IntexSpa/heater on
    payload_on: "1"
    payload_off: "0"
    name: "Intex Pool Heat"

  - unique_id: intex_pool_water_jet_switch
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd water jet on off
    state_topic: IntexSpa/Water jet on
    payload_on: "1"
    payload_off: "0"
    name: "Intex Pool Water Jet"

  - unique_id: intex_pool_sanitizer_switch
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    command_topic: IntexSpa/Cmd sanitizer on off
    state_topic: IntexSpa/Sanitizer on
    payload_on: "1"
    payload_off: "0"
    name: "Intex Pool Sanitizer"

  # MQTT Pool Sensoren
  sensor:
  - unique_id: intex_pool_status
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    name: "Intex Pool Status"
    state_topic: "IntexSpa/Error Number"
    unit_of_measurement: 'E'
  - unique_id: intex_pool_actual_temperature
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    name: "Intex Pool Actual Temperature"
    state_topic: "IntexSpa/Actual Temperature"
    unit_of_measurement: '°c'
  - unique_id: intex_pool_set_temperature
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    name: "Intex Pool Set Temperature"
    state_topic: "IntexSpa/Temperature Setpoint"
    unit_of_measurement: '°c'
  - unique_id: intex_pool_heater_status
    device:
      identifiers:
        - "Intex Spa"
      manufacturer: "Intex"
      model: "SPA"
      name: "Intex Spa Pool"
    name: "Intex Pool Heater Status"
    state_topic: "IntexSpa/heater state"
    value_template: >
      {% if value|float == 0 %}
        off
      {% elif value|float == 1 %}
        standby
      {% elif value|float == 2 %}
        on
      {% endif %}

  # Intex Pool Binärsensoren
  binary_sensor:
    - unique_id: intex_pool_pumpenkommunikation
      device:
        identifiers:
          - "Intex Spa"
        manufacturer: "Intex"
        model: "SPA"
        name: "Intex Spa Pool"
      name: "Intex Pool Pumpenkommunikation"
      state_topic: "IntexSpa/Communication with pump"
      payload_on: "1"
      payload_off: "0"
      device_class: connectivity
`

Portos1327, 2023-06-14:
hi rommess, thank you very much for your work I will look at what it looks like in Ha