Yogui79/IntexPureSpa #51: Channel Search

aliessum, 2022-06-14:
### Discussed in https://github.com/Yogui79/IntexPureSpa/discussions/20

<div type='discussions-op-text'>

<sup>Originally posted by **matthias811** July 20, 2021</sup>
First, thanks for providing the manual and code - I hope I will get this working!
Nevertheless I am stuck on Step 1, the channel search. The light is blinking continously and console is showing me all the channel numbers but I cannot detect a flickering like shown in your video. What is the distance the LC12s should be away from the pool? I tried a few meters away. Must there be something activated on the pool or is it just ok to have it on power with pump running?
I have ordered two LC12s by different accounts (to ensure one is there in time) and tried now with both, same effect....</div>

aliessum, 2022-06-14:
First of course many thanks for the effort you have put in the design!
I am facing the same problem as described in #20.
My spa is a Intex type #2845800A, according to the carton box. I have a LC12S V2.03 module (so not the blue one, but red/light brown like on your picture). ESP32 is a devkit C V4, so the one with the extending antenna board. 
During channel search the blue led is flashing slow and in the console I can see the numbers counting up. Distance to the pump-unit is less than 1 mtr, but a channel is not detected.
Is there a way that I can check the functionality of the LC12S (perhaps with an oscilloscope, I have one)?
Have you had experiences with bad LC12S modules, should I maybe get another LC12S?
Thanks in advance, Arno.

sjorsdeleede, 2022-07-16:
same problem here...
I tryed it with a Arduino Uno and with a ESP32
From channel 0 till channel FF there is no faster blinking light on any channel 
My spa is a 28462 from 2021, i tryed it with the pump running, i also tried with everything turned off and the LC12S was lying on top of the pump

funnyfrish, 2023-03-14:
Any solutions for that problem? How did you fix it?