Yogui79/IntexPureSpa #56: SearchChannel

MathieuJouslin, 2022-07-03:
Hello, 
I have a problem with the channel detection. The led is blinking the same way from channel 0 to FF. I change the delay in the loop function and nothing changed. I try with an arduino Uno, with an ESP32 and 2 different LC12S.
Someone have an idee ?

MathieuJouslin, 2022-07-04:
> Hello, I have a problem with the channel detection. The led is blinking the same way from channel 0 to FF. I change the delay in the loop function and nothing changed. I try with an arduino Uno, with an ESP32 and 2 different LC12S. Someone have an idee ?

Update, it works for me in the dark, the flashing was invisible because of the too bright environment.

dwavrant, 2022-08-07:
Hello @MathieuJouslin 

after Channel detection, did you have a quick blink with the correct channel for network id searching ?

Thanks

MathieuJouslin, 2022-08-11:
Hello @dwavrant, 
Yes after the channel detection i had a quick blink with the first channel (I had the 33 and 34 so I choose the first one).

Yogui79, 2022-08-13:
Hello 
could you please share your code