Yogui79/IntexPureSpa #62: The 2021 pool seems to use a different communication profile

PLCHome, 2023-03-05:
Hi, to shorten the search for the profile, I connected a TTL serial converter to the pool.

[image]

[image]

### That was the easy part

But the data looks strange:
[image]

```
00 AA 5A 00 00 AA 2C 00 00 00 04 00 2C 00 00 00 12 00 1C 
   1  2  3  4  5  6  7  8  9  10 11 12 13 14 15 16 17 18 
   
Self ID = 0000
NET ID = AA2C
Power = 0 12 dbm
Baud = 04 9600bt
Chanel = 2C
Length = 12   
   
   AA 5A 00 00 AA 2C 00 00 00 04 00 2C 00 00 00 12 00 1C 
   
   01 02 55 AA 03 0E AA BA
   01 02 55 AA 0B 01 0B 00 00 00 00 00 00 00 00 16 
   01 02 55 AA 0B 01 0B 00 00 00 00 00 00 00 00 16 
   01 02 55 AA 03 0E AA BA 
   01 02 55 AA 0B 01 0B 00 00 00 00 00 00 00 00 16 
   01 02 55 AA 0B 01 0B 02 02 00 FF FF 00 00 00 18 
   01 02 55 AA 0B 01 0B 02 02 00 00 00 00 00 00 1A 
   01 02 55 AA 0B 01 0B 02 02 00 00 00 00 00 00 1A 
   01 02 55 AA 02 0C 0D 
   01 02 55 AA 0B 01 0B 02 02 00 00 00 00 00 00 1A 
```

If I take the ESP I get the same data:
```
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA Debug Send command : 2C 00 00 00 00 00 3E EB  
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 01 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA Debug Send command : 2C 00 00 00 00 00 3E EB  
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA 
01 02 55 AA 0B 01 0B 02 02 00 00 03 00 00 00 1D 
01 02 55 AA 03 0E AA BA
```

That seems to be the length.
[image]

But I guess with the pool everything starts from scratch. They have changed the protocol with the Internet.

Do you have any idea?

Yogui79, 2023-03-06:
Hallo,

honestly I don't know if they changed the protocol, but in vaccording the trace you do louks like.
you seem to have the skills to trace l this, do you have any programming competences?
I will check your trace more detaillaed during the week and will try to provide you a small test programm to be able to check all information

could you just give me the pool model you have ?

coud ytou please poste the code you use for the trace ?

PLCHome, 2023-03-06:
It's an 28462 from 2021.
The trace is from spa.ino. I only changed the debug switches and then wrapped them in the text program at 01 02 55 AA.
I increse the timeout time too.

My C has some rust on it. 
Last summer I had adjusted the code for reading via Wifi for the ioBroker.
But the wifi protocol is really bad. Far too slow and cumbersome.

I was hoping this one would work better.
So second mistake, first sold the old one because I thought it would work better with Wifi (OK the old one was too small too) then I have determined that Wifi sucks. And now this doesn't work anymore either.
That's luck...

Yogui79, 2023-03-06:
I think it's possible to also read the information from your spa
If you want I can also provide a code for Arduino mega and and a w5100 ethernet shield

Yogui79, 2023-03-06:
could you please provide log with followind #define configuration

`#define DEBUG_RECIEVED_DATA
//#define DEBUG_SEARCH_CHANNEL
//#define DEBUG_SEND_COMMAND
//#define DEBUG_PUMP_DATA
//#define DEBUG_CONTROLLER_DATA
//#define DEBUG_CONFIG
//#define DEBUG_MQTT
//#define DEBUG_SEND_VALUE_TO_HOME_AUTOMATION_SW`