Yogui79/IntexPureSpa #78: No channel detection - version of LC12s compatibility ?

courtoiss, 2025-05-21:
Hello
I've got a spa SC20 and i try to connect it form an esp32 + LC12s as you do.

I can't detect the channel. The led is blinking normaly, but no detection as you show in your video.
I've read many of the messages posted here, and double check my connections. I think i'm good. 
I've tried near and far, with pump on and off, etc... 
I've also tried the modified scripts by @fsedarkalex , without luck.

Do you have some advice ? Can i check some logs to be sure my connections are good ?

Moreover, I have seein in the recent messages that people have the same problem. Do you think it's possible that is the LC12s current version wich is not compatible ? I have recived the LC12s v2.09

I read on 
https://fr.aliexpress.com/item/32823346122.html?algo_exp_id=ded99d33-2f1b-4169-a0b3-b6aa9c5d8924-12&pdp_ext_f=%7B%22order%22%3A%221%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis!EUR!1.29!1.29!!!1.42!1.42!%402103846917478436976164031e181d!66701546982!sea!FR!0!ABX&curPageLogUid=HjY8rGqzPuIh&utparam-url=scene%3Asearch%7Cquery_from%3A#nav-specification
that some models can be not compatible.

Do you know if i must buy another LC12s card ? wich version minimum - maximum ? What is YOUR version ? 
Do you know some vendor that have old version to sell ?

Maybe i'm confusing. If you have some other advice, thank you.
Best regards
Sylvain (France)

alexh-fr, 2025-06-04:
> PS: i'm now waiting for the NetworkID, I cross the fingers.

Looking forward to reading your results!
I also bought an "older" model of LC12S on Aliexpress, but the package arrived empty...

I'm working around my need by resuming development of an alternative to control my spa using Python (but the spa must be connected to the Wi-Fi network).

fsedarkalex, 2025-06-29:
Ich glaube die WLAN Variante hatte ein anderes Kommunikationsmodul. Ich bin nicht 100% sicher aber ich glaube die ist nicht kompatibel Sent from NineVon: ghost7203 ***@***.***>Gesendet: Sonntag, 29. Juni 2025 05:50An: Yogui79/IntexPureSpaCc: fsedarkalex; MentionBetreff: Re: [Yogui79/IntexPureSpa] No channel detection - version of LC12s compatibility ? (Issue #78) ghost7203 left a comment (Yogui79/IntexPureSpa#78)

Hallo Alles ist in Ordnung, es funktioniert! Vielen Dank für die Arbeit und die Dokumentation. Ich habe alle Zustände in meinem Jeedom enthalten und Befehle erstellt, um die Aktionen zu starten. Ich denke, Sie müssen sich entscheiden, ob Sie sich mit dem lc12s oder über WLAN verbinden möchten. Das hängt von der Art des Spas ab. Ein Spa kann nicht beides.

Ich habe auch das Thema das ich weder Channel noch netzwerk id richtig finde. mein Spa ist mit dem wlan und der App verbunden muss ich das zuerst trennen bevor ich scanne ? Wäre über Hilfe unendlich dankbar. ich habe einen Intex pur spa 28442-018 /20210423 JPA20

[
{
"target": "https://github.com/Yogui79/IntexPureSpa/issues/78#issuecomment-3016273043",
"url": "https://github.com/Yogui79/IntexPureSpa/issues/78#issuecomment-3016273043",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

ghost7203, 2025-06-30:
Hallo Fsedarkalex, 
Nach dem ich ausgiebig getestet habe mit dem Channel hab ich ein ganz schwaches blinken auf Kanal 59 am LC12 endteckt, also glaube ich das ich schon einen LC12 in der Poolsteuerung verbaut habe, es ist wirklich ein ganz ganz schwaches blinken fast nicht sichtbar. Was meinst du als Experte dazu? jetzt versuche ich die Netzwerk ID rauszufinden, wie sollte das aussehen wenn ich mitloge via Putty, kannst du mir da so ein Besispiel bild senden.
Vielen lieben Dank und Grüße aus Vienna Robert

fsedarkalex, 2025-06-30:
Du siehst an der Konsole nichts davon, weil die network id nicht stimmt.Aber das minimale Blinken ist richtig. MN sieht es nur schwer.Sent from NineVon: ghost7203 ***@***.***>Gesendet: Montag, 30. Juni 2025 06:31An: Yogui79/IntexPureSpaCc: fsedarkalex; MentionBetreff: Re: [Yogui79/IntexPureSpa] No channel detection - version of LC12s compatibility ? (Issue #78) ghost7203 left a comment (Yogui79/IntexPureSpa#78)
Hallo Fsedarkalex,
Nach dem ich ausgiebig getestet habe mit dem Channel hab ich ein ganz schwaches blinken auf Kanal 59 am LC12 endteckt, also glaube ich das ich schon einen LC12 in der Poolsteuerung verbaut habe, es ist wirklich ein ganz ganz schwaches blinken fast nicht sichtbar. Was meinst du als Experte dazu? jetzt versuche ich die Netzwerk ID rauszufinden, wie sollte das aussehen wenn ich mitloge via Putty, kannst du mir da so ein Besispiel bild senden.
Vielen lieben Dank und Grüße aus Vienna Robert

[
{
"target": "https://github.com/Yogui79/IntexPureSpa/issues/78#issuecomment-3017745369",
"url": "https://github.com/Yogui79/IntexPureSpa/issues/78#issuecomment-3017745369",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

ghost7203, 2025-06-30:
Ich meinte wenn ich die Netzwerk ID im zweiten schritt mache und diese Aufzeichne kommen pr scan im Log 2 Einträge 
config AA 00 .. .. ... ...
AA  00  ..  .. ..
Wie sollte es ausehen wenn er die Netzwerk ID  findet (wie ist dann der Eintrag)
Danke 
LG Robert

mcmjau, 2026-06-07:
Bonjour, j'ai réussi a trouver le channel et le network en commandant un nouveau LC12s (v2.09) sur ali, j'arrive à récupérer les infos du spa sur MQTT par contre impossible d'envoyer un ordre, j'ai vérifié mon câblage et mes soudures et la je sèche... Quelqu'un aurait une idée ?